Decawave/dwm1001-examples #3: Unable to use GPIO 10

LeyreMF, 2018-05-30:
I would like to use GPIO10 for an extra LED but I can't set it high to light the led.
Is there any issue in the module in this pin?

dmolteni, 2018-05-30:
Hi,
I have similar problem but using Gpio14 to control led D12...

YBDecawave, 2018-05-31:
Hi dmolteni, 

There should be no issue controlling D12 with the GPIO14. You can use the mask BSP_LED_3_MASK which effectively maps to the GPIO14.

LEDS_CONFIGURE(BSP_LED_3_MASK);
LEDS_ON(BSP_LED_3_MASK);
LEDS_OFF(BSP_LED_3_MASK);

LeyreMF, I'm looking at the gpio10 and come back to you.

Thanks
Regards
Yves

LeyreMF, 2018-05-31:
Thank you in advance, I have tested with other pins such as GPIO31 with my custom function task_GPIO(PIN,WRITE_1) and It works. I don't know why GPIO10 is unconfigurable.

dmolteni, 2018-05-31:
Hi,
thanks!!

Davide

YBDecawave, 2018-06-01:
Hi LeyreMF,

With the nRF52832, the GPIOs 10 & 9 are reserved for NFC communication by default. It is possible use them as GPIOs but you need to define the following pre-processor at compilation : CONFIG_NFCT_PINS_AS_GPIOS.

To add this preprocessor in SES, right click on solution -> edit options -> preprocessor -> add "CONFIG_NFCT_PINS_AS_GPIOS" in "preprocessor definition"

It will free GPIOs 10 & 9 and you will be able to toggle them as classical GPIOs. I've tried on my side and it works. 

There is more infromation regarding the NFC and GPIOs 9 & 10 on the NRF website : 

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.nrf52832.ps.v1.1%2Fnfc.html&cp=2_1_0_41_1&anchor=concept_fq1_1bk_1s

Let me how it goes.
Thank you,
Regards
Yves

LeyreMF, 2018-06-01:
Thank you very much!!!
I'll try it!!

LeyreMF, 2018-06-04:
Thank you very much! It works perfectly. Now, both pins can be configurated!

LeyreMF, 2018-06-04:
One question more, have you tested the RTLS library provided by Decawave in the VirtualBox? I tried to use it in combination with other sensor connected in SPIM2(DWM1001 is connected in SPIM1 and the accelerometer in TWI0) but It was impossible to incorporate it.

closed 2019-03-12 by YBDecawave