Decawave/dwm1001-examples #6: Unable to use the  NRF_LOG_INFO()

lddssxy, 2018-07-03:
Hi, I am using DWM1001 to try to run the example code. Once I enable NRF_LOG and NRF_LOG_BACKEND_UART in sdk_config.h, The program cannot run properly. After I disabled NRF_LOG and NRF_LOG_BACKEND_UART, everything goes all right. 
Is this the problem of the module?  I have tried also the examples in the Nordic nRF5_SDK_14.2.0. The problem is the same, I cannot read anything out from the UART, because Nordic examples are using NRF_LOG_INFO() to implement the UART.

lddssxy, 2018-07-04:
If I can change some setting to use NRF_LOG_INFO() to print out to the UART, or I just cannot use this function with DWM1001?

YBDecawave, 2018-07-04:
Hi lddsxy,

I'll have to take a look at that, but what do you need to use the backend log ? Printf is already redirected on the uart so it should be enough no ?

Thanks
Yves

lddssxy, 2018-07-04:
Hi Yves,
Thank you for your reply:) 

As you said, the printf directed on UART is good.  But because the example in the SDK is using the backend log on UART, I really want to try to run those examples properly which are good base for me to develop the BLE part in DWM1001 module.

alternatively, could you please give me some hint of how to redirected the printf on the UART?
The example I am trying is in Nordic nRF5_SDK_14.2.0, the file path is 
nRF5_SDK_14.2.0\examples\ble_central\ble_app_hrs_c\pca10040\s132\ses\ble_app_hrs_c_pca10040_s132.

Best regards
Yu

lddssxy, 2018-07-05:
Hi Yves,
I think I got some idea to solve this problem. the reason I want use the NRF_LOG_INFO() with DWM1001 is that the examples in Nordic SDK using this function to debug. I can use  the Real Time Terminal(RTT) from SEGGER embedded studio. 
Following link gives good instruction about how to use the RTT:
https://devzone.nordicsemi.com/tutorials/b/getting-started/posts/debugging-with-real-time-terminal

closed 2018-09-20 by lddssxy