Decawave/dwm1001-examples #15: Error while building ss_twr_init

asmitajha, 2019-06-26:
Hello,
I have successfully built ss_twr_resp. I am getting error while building ss_twr_init. I have send the log here , there is controversy in _putchar.
Building ‘ss_twr_init’ from solution ‘ss_twr_init’ in configuration ‘nrf52832_xxaa’
  Compiling ‘retarget.c’
    "C:/Program Files/SEGGER/SEGGER Embedded Studio for ARM 4.16/gcc/arm-none-eabi/bin/cc1" -fmessage-length=0 -fno-diagnostics-show-caret -mcpu=cortex-m4 -mlittle-endian -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mtp=soft -munaligned-access -nostdinc "-isystemC:/Program Files/SEGGER/SEGGER Embedded Studio for ARM 4.16/include" "-isystemC:/Users/Admin/AppData/Local/SEGGER/SEGGER Embedded Studio/v3/packages/include" "-IC:/Users/Admin/AppData/Local/SEGGER/SEGGER Embedded Studio/v3/packages/nRF/CMSIS/Device/Include" "-IC:/Users/Admin/AppData/Local/SEGGER/SEGGER Embedded Studio/v3/packages/CMSIS_5/CMSIS/Core/Include" -I../../../nRF5_SDK_14.2.0/components/device -I../UART -I../config -I../../../nRF5_SDK_14.2.0/components -I../../../nRF5_SDK_14.2.0/components/boards -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/clock -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/common -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/delay -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/gpiote -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/hal -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/nrf_soc_nosd -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/spi_master -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/uart -I../../../nRF5_SDK_14.2.0/components/libraries/atomic -I../../../nRF5_SDK_14.2.0/components/libraries/balloc -I../../../nRF5_SDK_14.2.0/components/libraries/bsp -I../../../nRF5_SDK_14.2.0/components/libraries/button -I../../../nRF5_SDK_14.2.0/components/libraries/fifo -I../../../nRF5_SDK_14.2.0/components/libraries/experimental_log -I../../../nRF5_SDK_14.2.0/components/libraries/experimental_log/src -I../../../nRF5_SDK_14.2.0/components/libraries/experimental_memobj -I../../../nRF5_SDK_14.2.0/components/libraries/experimental_section_vars -I../../../nRF5_SDK_14.2.0/components/libraries/scheduler -I../../../nRF5_SDK_14.2.0/components/libraries/strerror -I../../../nRF5_SDK_14.2.0/components/libraries/timer -I../../../nRF5_SDK_14.2.0/components/libraries/util -I../../../nRF5_SDK_14.2.0/components/libraries/uart -I../../../nRF5_SDK_14.2.0/components/toolchain -I../../../deca_driver -I../../../deca_driver/port -I../../../nRF5_SDK_14.2.0/external/fprintf -I../../../nRF5_SDK_14.2.0/external/segger_rtt -I../../../nRF5_SDK_14.2.0/external/freertos/ -I../../../nRF5_SDK_14.2.0/external/freertos/source -I../../../nRF5_SDK_14.2.0/external/freertos/config -I../../../nRF5_SDK_14.2.0/external/freertos/source/include -I../../../nRF5_SDK_14.2.0/external/freertos/portable/ARM/nrf52 -I../../../nRF5_SDK_14.2.0/external/freertos/portable/CMSIS/nrf52 -I../../../nRF5_SDK_14.2.0/external/freertos/source/portable -I../../../boards -I../../.. -D__SIZEOF_WCHAR_T=4 -D__ARM_ARCH_7EM__ -D__SES_ARM -D__ARM_ARCH_FPV4_SP_D16__ -D__SES_VERSION=41600 -DBOARD_DW1001_DEV -DBSP_SIMPLE -DCONFIG_GPIO_AS_PINRESET -DFLOAT_ABI_HARD -DNRF52 -DNRF52832_XXAA -DNRF52_PAN_74 -DSWI_DISABLE0 -MD "E:/Projects/RTLS/nRF/GitCodes/dwm1001-examples/dwm1001-examples/examples/ss_twr_init/SES/Output/ss_twr_init nrf52832_xxaa/Obj/retarget.d" -MQ "Output/ss_twr_init nrf52832_xxaa/Obj/retarget.o" -quiet -std=gnu99 -g2 -gpubnames -fomit-frame-pointer -fno-dwarf2-cfi-asm -fno-builtin -ffunction-sections -fdata-sections -fshort-enums -fno-common E:\Projects\RTLS\nRF\GitCodes\dwm1001-examples\dwm1001-examples\nRF5_SDK_14.2.0\components\libraries\uart\retarget.c -o "E:/Projects/RTLS/nRF/GitCodes/dwm1001-examples/dwm1001-examples/examples/ss_twr_init/SES/Output/ss_twr_init nrf52832_xxaa/Obj/retarget.asm"
    conflicting types for '__putchar'
    retarget.c
    previous declaration of '__putchar' was here
  Compiling ‘nrf_log_backend_rtt.c’
    "C:/Program Files/SEGGER/SEGGER Embedded Studio for ARM 4.16/gcc/arm-none-eabi/bin/cc1" -fmessage-length=0 -fno-diagnostics-show-caret -mcpu=cortex-m4 -mlittle-endian -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mtp=soft -munaligned-access -nostdinc "-isystemC:/Program Files/SEGGER/SEGGER Embedded Studio for ARM 4.16/include" "-isystemC:/Users/Admin/AppData/Local/SEGGER/SEGGER Embedded Studio/v3/packages/include" "-IC:/Users/Admin/AppData/Local/SEGGER/SEGGER Embedded Studio/v3/packages/nRF/CMSIS/Device/Include" "-IC:/Users/Admin/AppData/Local/SEGGER/SEGGER Embedded Studio/v3/packages/CMSIS_5/CMSIS/Core/Include" -I../../../nRF5_SDK_14.2.0/components/device -I../UART -I../config -I../../../nRF5_SDK_14.2.0/components -I../../../nRF5_SDK_14.2.0/components/boards -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/clock -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/common -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/delay -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/gpiote -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/hal -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/nrf_soc_nosd -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/spi_master -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/uart -I../../../nRF5_SDK_14.2.0/components/libraries/atomic -I../../../nRF5_SDK_14.2.0/components/libraries/balloc -I../../../nRF5_SDK_14.2.0/components/libraries/bsp -I../../../nRF5_SDK_14.2.0/components/libraries/button -I../../../nRF5_SDK_14.2.0/components/libraries/fifo -I../../../nRF5_SDK_14.2.0/components/libraries/experimental_log -I../../../nRF5_SDK_14.2.0/components/libraries/experimental_log/src -I../../../nRF5_SDK_14.2.0/components/libraries/experimental_memobj -I../../../nRF5_SDK_14.2.0/components/libraries/experimental_section_vars -I../../../nRF5_SDK_14.2.0/components/libraries/scheduler -I../../../nRF5_SDK_14.2.0/components/libraries/strerror -I../../../nRF5_SDK_14.2.0/components/libraries/timer -I../../../nRF5_SDK_14.2.0/components/libraries/util -I../../../nRF5_SDK_14.2.0/components/libraries/uart -I../../../nRF5_SDK_14.2.0/components/toolchain -I../../../deca_driver -I../../../deca_driver/port -I../../../nRF5_SDK_14.2.0/external/fprintf -I../../../nRF5_SDK_14.2.0/external/segger_rtt -I../../../nRF5_SDK_14.2.0/external/freertos/ -I../../../nRF5_SDK_14.2.0/external/freertos/source -I../../../nRF5_SDK_14.2.0/external/freertos/config -I../../../nRF5_SDK_14.2.0/external/freertos/source/include -I../../../nRF5_SDK_14.2.0/external/freertos/portable/ARM/nrf52 -I../../../nRF5_SDK_14.2.0/external/freertos/portable/CMSIS/nrf52 -I../../../nRF5_SDK_14.2.0/external/freertos/source/portable -I../../../boards -I../../.. -D__SIZEOF_WCHAR_T=4 -D__ARM_ARCH_7EM__ -D__SES_ARM -D__ARM_ARCH_FPV4_SP_D16__ -D__SES_VERSION=41600 -DBOARD_DW1001_DEV -DBSP_SIMPLE -DCONFIG_GPIO_AS_PINRESET -DFLOAT_ABI_HARD -DNRF52 -DNRF52832_XXAA -DNRF52_PAN_74 -DSWI_DISABLE0 -MD "E:/Projects/RTLS/nRF/GitCodes/dwm1001-examples/dwm1001-examples/examples/ss_twr_init/SES/Output/ss_twr_init nrf52832_xxaa/Obj/nrf_log_backend_rtt.d" -MQ "Output/ss_twr_init nrf52832_xxaa/Obj/nrf_log_backend_rtt.o" -quiet -std=gnu99 -g2 -gpubnames -fomit-frame-pointer -fno-dwarf2-cfi-asm -fno-builtin -ffunction-sections -fdata-sections -fshort-enums -fno-common E:\Projects\RTLS\nRF\GitCodes\dwm1001-examples\dwm1001-examples\nRF5_SDK_14.2.0\components\libraries\experimental_log\src\nrf_log_backend_rtt.c -o "E:/Projects/RTLS/nRF/GitCodes/dwm1001-examples/dwm1001-examples/examples/ss_twr_init/SES/Output/ss_twr_init nrf52832_xxaa/Obj/nrf_log_backend_rtt.asm"
  Compiling ‘nrf_log_backend_serial.c’
    "C:/Program Files/SEGGER/SEGGER Embedded Studio for ARM 4.16/gcc/arm-none-eabi/bin/cc1" -fmessage-length=0 -fno-diagnostics-show-caret -mcpu=cortex-m4 -mlittle-endian -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mthumb -mtp=soft -munaligned-access -nostdinc "-isystemC:/Program Files/SEGGER/SEGGER Embedded Studio for ARM 4.16/include" "-isystemC:/Users/Admin/AppData/Local/SEGGER/SEGGER Embedded Studio/v3/packages/include" "-IC:/Users/Admin/AppData/Local/SEGGER/SEGGER Embedded Studio/v3/packages/nRF/CMSIS/Device/Include" "-IC:/Users/Admin/AppData/Local/SEGGER/SEGGER Embedded Studio/v3/packages/CMSIS_5/CMSIS/Core/Include" -I../../../nRF5_SDK_14.2.0/components/device -I../UART -I../config -I../../../nRF5_SDK_14.2.0/components -I../../../nRF5_SDK_14.2.0/components/boards -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/clock -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/common -I../../../nRF5_SDK_14.2.0/components/drivers_nrf/delay -I../../../nRF5_SDK_14.2.0/components/dri [... cut ...] art.o" -quiet -std=gnu99 -g2 -gpubnames -fomit-frame-pointer -fno-dwarf2-cfi-asm -fno-builtin -ffunction-sections -fdata-sections -fshort-enums -fno-common E:\Projects\RTLS\nRF\GitCodes\dwm1001-examples\dwm1001-examples\nRF5_SDK_14.2.0\components\libraries\experimental_log\src\nrf_log_backend_uart.c -o "E:/Projects/RTLS/nRF/GitCodes/dwm1001-examples/dwm1001-examples/examples/ss_twr_init/SES/Output/ss_twr_init nrf52832_xxaa/Obj/nrf_log_backend_uart.asm"
Build failed

Kindly suggest

Thanks
Asmita

RogetK, 2019-07-02:
Hi Asmita, 

This bug is a known problem in the project due to updates to Segger Embedded Studio causing conflicting types, if you pull from my fork of the project, you should have no problems at all.

jonathanrjpereira, 2020-07-07:
@asmitajha  @RogetK  I got a similar error:

```
Building ‘ss_twr_init’ from solution ‘ss_twr_init’ in configuration ‘nrf52832_xxaa’
  Compiling ‘retarget.c’
    conflicting types for '__putchar'
    retarget.c
    previous declaration of '__putchar' was here
  Compiling ‘nrf_log_backend_rtt.c’
  Compiling ‘app_uart_fifo.c’
  Compiling ‘nrf_log_backend_serial.c’
Build failed
```