Decawave/dwm1001-examples #29: Error: unknown type name '__printf_tag_ptr'

nuradlinn, 2022-04-08:
Hi there,

I want to build ss_twr_init, but I kept getting an error. I have followed the changes as shown in the pull requests, but I still kept getting the same error.
Error: unknown type name '__printf_tag_ptr'

Do you have any suggestions on how can I fix this?

Thank you

Maxwh21, 2022-04-15:
I'm having the same issue, did you manage to solve it?

nuradlinn, 2022-04-18:
Unfortunately, I did not manage to solve the error. But since I just want to make my DWM1001-DEV run first, I followed the Decawave's Gateway Quick Deployment Guide. Then, I manage to set the UWB into tag and anchors.

Maxwh21, 2022-04-18:
Nice cheers I have managed to upload some code using the 'firmware user guide' but I am struggling to find any meaningful output I will try the quick deployment guide next.

Maxwh21, 2022-04-25:
> Unfortunately, I did not manage to solve the error. But since I just want to make my DWM1001-DEV run first, I followed the Decawave's Gateway Quick Deployment Guide. Then, I manage to set the UWB into tag and anchors.

I have followed the gateway quick development guide and managed to get the server running on the Pi, but I cannot locate the  tags and anchors within this server. I am assuming this is because I skipped a step (downloading the android application) so I will do this next. My question to you is after you had completed the gateway guide, did you go back to using Segger? I am unsure of what the use of Segger is if its not needed in the gateway guide

youssef-itanii, 2023-07-07:
Changing the Library I/O allowed compilation without errors for me. However, I can't seem to obtain any output text when connecting on a serial terminal (PuTTY)