Decawave/dwm1001-examples #30: Linker error: Undefined reference to `stdout`

dslima90, 2022-04-30:
The code compiles but i am getting some linker error. The log is bellow:

```
2> Generating linker script 'ss_twr_init.ld'[0m
2> Linking ss_twr_init.elf[0m
2> /usr/share/segger_embedded_studio_for_arm_6.22a/gcc/arm-none-eabi/bin/ld: /usr/share/segger_embedded_studio_for_arm_6.22a/lib/libc_v7em_fpv4_sp_d16_hard_t_le_eabi_balanced.a(fileops.o): in function `putchar':[0m
2> fileops.c:(.text.libc.putchar+0x0): undefined reference to `stdout'[0m
2> /usr/share/segger_embedded_studio_for_arm_6.22a/gcc/arm-none-eabi/bin/ld: fileops.c:(.text.libc.putchar+0x4): undefined reference to `stdout'[0m
2> /usr/share/segger_embedded_studio_for_arm_6.22a/gcc/arm-none-eabi/bin/ld: /usr/share/segger_embedded_studio_for_arm_6.22a/lib/libc_v7em_fpv4_sp_d16_hard_t_le_eabi_balanced.a(fileops.o): in function `puts':[0m
2> fileops.c:(.text.libc.puts+0x2): undefined reference to `stdout'[0m
2> /usr/share/segger_embedded_studio_for_arm_6.22a/gcc/arm-none-eabi/bin/ld: fileops.c:(.text.libc.puts+0x6): undefined reference to `stdout'[0m
2> /usr/share/segger_embedded_studio_for_arm_6.22a/gcc/arm-none-eabi/bin/ld: fileops.c:(.text.libc.puts+0x18): undefined reference to `__SEGGER_RTL_X_file_write'[0m
2> /usr/share/segger_embedded_studio_for_arm_6.22a/gcc/arm-none-eabi/bin/ld: /usr/share/segger_embedded_studio_for_arm_6.22a/lib/libc_v7em_fpv4_sp_d16_hard_t_le_eabi_balanced.a(fileops.o): in function `fwrite':[0m
2> fileops.c:(.text.libc.fwrite+0xc): undefined reference to `__SEGGER_RTL_X_file_stat'[0m
2> /usr/share/segger_embedded_studio_for_arm_6.22a/gcc/arm-none-eabi/bin/ld: fileops.c:(.text.libc.fwrite+0x24): undefined reference to `__SEGGER_RTL_X_file_write'[0m
2> /usr/share/segger_embedded_studio_for_arm_6.22a/gcc/arm-none-eabi/bin/ld: /usr/share/segger_embedded_studio_for_arm_6.22a/lib/libc_v7em_fpv4_sp_d16_hard_t_le_eabi_balanced.a(prinops.o): in function `vfprintf':[0m
2> prinops.c:(.text.libc.vfprintf+0xe): undefined reference to `__SEGGER_RTL_X_file_bufsize'[0m
2> /usr/share/segger_embedded_studio_for_arm_6.22a/gcc/arm-none-eabi/bin/ld: /usr/share/segger_embedded_studio_for_arm_6.22a/lib/libc_v7em_fpv4_sp_d16_hard_t_le_eabi_balanced.a(prinops.o): in function `printf':[0m
2> prinops.c:(.text.libc.printf+0xe): undefined reference to `stdout'[0m
2> /usr/share/segger_embedded_studio_for_arm_6.22a/gcc/arm-none-eabi/bin/ld: prinops.c:(.text.libc.printf+0x12): undefined reference to `stdout'[0m
Build failed[0m

```

Can anyone help-me?

O downloaded the most recent version of segger studio and installed the necessary packages

```
SEGGER Embedded Studio for ARM
Release 6.22a Build 2022032300.49847
Linux x64
```

giopoli2000, 2022-05-27:
Hello, did you happen to solve the issue on this matter? im obtaining the same error on the same version of Segger Embedded, I would like to make this project work for this particular new version of segger

giopoli2000, 2022-05-27:
But if you want the easy fix, just download the Segger Embedded version 5.70a and  you will not find the same linker problems you are having now

dslima90, 2022-05-27:
Hello, @giopoli2000 . I could not solve the problem, 'solved' by downloading an old version of Segger Embedded and it is now working. 

It would be great if this project could be upgraded for the new version of Segger Embedded!

Thank you for the reply!

mattcaves, 2022-05-29:
Hi all, 
I had this problem too with the latest SES (6.30) version, after a bit of searching I found this post on the Nordic forums that helped solve it:
[https://devzone.nordicsemi.com/f/nordic-q-a/86110/nrf5-sdk-17-1-0-examples-is-not-compiling-in-latest-ses-6-20a](https://devzone.nordicsemi.com/f/nordic-q-a/86110/nrf5-sdk-17-1-0-examples-is-not-compiling-in-latest-ses-6-20a
)
Cheers,
Matt

hongthiet1994, 2022-06-14:
Thank you so much