Decawave/dwm1001-examples #32: build failed in segger embedded studio

lokeycookie, 2024-07-09:
Hi there,
I am trying to flash the initiator C program into my DWM1001-dev board but I got the below error while trying to build the C program in segger embedded studio.
How do I solve this?

Checking project status
Building 'ss_twr_init' from solution 'ss_twr_init' in configuration 'nrf52832_xxaa'
  Checking 'main.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/main.o' is up to date
  Checking 'ss_init_main.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/ss_init_main.o' is up to date
  Checking 'boards.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/boards.o' is up to date
  Checking 'bsp.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/bsp.o' is up to date
  Checking 'bsp_nfc.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/bsp_nfc.o' is up to date
  Checking 'deca_device.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/deca_device.o' is up to date
  Checking 'deca_params_init.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/deca_params_init.o' is up to date
  Checking 'port_platform.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/port_platform.o' is up to date
  Checking 'app_timer_freertos.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/app_timer_freertos.o' is up to date
  Checking 'croutine.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/croutine.o' is up to date
  Checking 'event_groups.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/event_groups.o' is up to date
  Checking 'heap_1.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/heap_1.o' is up to date
  Checking 'list.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/list.o' is up to date
  Checking 'port.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/port.o' is up to date
  Checking 'port_cmsis.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/port_cmsis.o' is up to date
  Checking 'port_cmsis_systick.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/port_cmsis_systick.o' is up to date
  Checking 'queue.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/queue.o' is up to date
  Checking 'tasks.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/tasks.o' is up to date
  Checking 'timers.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/timers.o' is up to date
  Checking 'ses_nrf52_startup.s'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/ses_nrf52_startup.o' is up to date
  Checking 'system_nrf52.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/system_nrf52.o' is up to date
  Checking 'thumb_crt0.s'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/thumb_crt0.o' is up to date
  Checking 'nrf_drv_clock.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_drv_clock.o' is up to date
  Checking 'nrf_drv_common.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_drv_common.o' is up to date
  Checking 'nrf_drv_gpiote.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_drv_gpiote.o' is up to date
  Checking 'nrf_drv_spi.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_drv_spi.o' is up to date
  Checking 'nrf_drv_uart.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_drv_uart.o' is up to date
  Checking 'app_button.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/app_button.o' is up to date
  Checking 'app_error.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/app_error.o' is up to date
  Checking 'app_error_weak.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/app_error_weak.o' is up to date
  Checking 'app_fifo.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/app_fifo.o' is up to date
  Checking 'app_scheduler.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/app_scheduler.o' is up to date
  Checking 'app_uart_fifo.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/app_uart_fifo.o' is up to date
  Checking 'app_util_platform.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/app_util_platform.o' is up to date
  Checking 'nrf_assert.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_assert.o' is up to date
  Checking 'nrf_balloc.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_balloc.o' is up to date
  Checking 'nrf_fprintf.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_fprintf.o' is up to date
  Checking 'nrf_fprintf_format.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_fprintf_format.o' is up to date
  Checking 'nrf_memobj.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_memobj.o' is up to date
  Checking 'nrf_strerror.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_strerror.o' is up to date
  Checking 'retarget.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/retarget.o' is up to date
  Checking 'nrf_log_backend_rtt.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_log_backend_rtt.o' is up to date
  Checking 'nrf_log_backend_serial.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_log_backend_serial.o' is up to date
  Checking 'nrf_log_backend_uart.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_log_backend_uart.o' is up to date
  Checking 'nrf_log_default_backends.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_log_default_backends.o' is up to date
  Checking 'nrf_log_frontend.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_log_frontend.o' is up to date
  Checking 'nrf_log_str_formatter.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/nrf_log_str_formatter.o' is up to date
  Checking 'SEGGER_RTT.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/SEGGER_RTT.o' is up to date
  Checking 'SEGGER_RTT_printf.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/SEGGER_RTT_printf.o' is up to date
  Checking 'SEGGER_RTT_Syscalls_KEIL.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/SEGGER_RTT_Syscalls_KEIL.o' is up to date
  Checking 'UART.c'
    'Output/nrf52832_xxaa/Obj/ss_twr_init/UART.o' is up to date
  Checking 'ss_twr_init'
    Output/nrf52832_xxaa/Exe/ss_twr_init.elf does not exist.
  Linking ss_twr_init.elf
     /opt/SEGGER/segger_embedded_studio_8.10d/lib/libc_v7em_fpv4_sp_d16_hard_t_le_eabi_balanced.a(fileops.o): in function `putchar':
    fileops.c:(.text.libc.putchar+0x0): undefined reference to `stdout'
     fileops.c:(.text.libc.putchar+0x4): undefined reference to `stdout'
     /opt/SEGGER/segger_embedded_studio_8.10d/lib/libc_v7em_fpv4_sp_d16_hard_t_le_eabi_balanced.a(fileops.o): in function `puts':
    fileops.c:(.text.libc.puts+0x2): undefined reference to `stdout'
     fileops.c:(.text.libc.puts+0x6): undefined reference to `stdout'
     fileops.c:(.text.libc.puts+0x18): undefined reference to `__SEGGER_RTL_X_file_write'
     /opt/SEGGER/segger_embedded_studio_8.10d/lib/libc_v7em_fpv4_sp_d16_hard_t_le_eabi_balanced.a(fileops.o): in function `fwrite':
    fileops.c:(.text.libc.fwrite+0xc): undefined reference to `__SEGGER_RTL_X_file_stat'
     fileops.c:(.text.libc.fwrite+0x24): undefined reference to `__SEGGER_RTL_X_file_write'
     /opt/SEGGER/segger_embedded_studio_8.10d/lib/libc_v7em_fpv4_sp_d16_hard_t_le_eabi_balanced.a(prinops.o): in function `vfprintf_l':
    prinops.c:(.text.libc.vfprintf_l+0x12): undefined reference to `__SEGGER_RTL_X_file_bufsize'
     /opt/SEGGER/segger_embedded_studio_8.10d/lib/libc_v7em_fpv4_sp_d16_hard_t_le_eabi_balanced.a(prinops.o): in function `printf':
    prinops.c:(.text.libc.printf+0xe): undefined reference to `stdout'
     prinops.c:(.text.libc.printf+0x12): undefined reference to `stdout'
    Build failed
Build failed

sidharthanup, 2024-12-13:
Hello! Did you get this resolved by any chance?

Muhammad-Haziq911, 2025-05-14:
I believe in the v6.20 release, Segger removed the Embedded Studio v4 runtime library from SES, which is the library with which nRF SDKs have been developed. From v6.20 and later, you are forced to use their [emRun](https://www.segger.com/products/development-tools/runtime-library/) library instead.
My suggestion is to use the SES v5.70, I have been using it and it works fine. 
https://devzone.nordicsemi.com/f/nordic-q-a/105785/library-heap-not-functional-when-migrating-from-ses-5-70a-to-ses-7-32